txoof/epdlib #1: Add support for colored screens

txoof, 2020-09-27:
3 color screens throw errors when used with this code

txoof, 2020-09-27:
simple fix is to change how `screen` handles displaying images -- pass an empty 1x1 image when dealing with a colored display.

This won't add color support, but will fix the problem of thrown errors.

txoof, 2020-09-29:
commit https://github.com/txoof/epdlib/commit/70b168360eee31a5c8f2376118332048796ad674 should resolve this with the added cost of much slower updates.

This can pave the way to adding color output.

txoof, 2020-10-01:
merge in suggestions from pull request #2

txoof, 2020-10-02:
Closed with commit https://github.com/txoof/epdlib/commit/7bf3865969cf1e1840cfd4878e220b7344609b5e

closed 2020-10-02 by txoof

tmmorin, 2020-10-06:
This fix works just fine, thank you !