txoof/epdlib #7: Wrong rotation by

tmmorin, 2020-10-12:
https://github.com/txoof/epdlib/blob/b1cbb878596aef58fdde4576f69cf4c599b04f6d/epdlib/Screen.py#L368-L369

This will not, I think, give the expected result for 90 degrees, nor for -90.

Since supported rotation were checked earlier, I think this should just be:
```
            image = image.rotate(self.rotation)
```

txoof, 2020-10-13:
This appears to work on my development setup. 

I've since merged these changes into the master branch. Have you tried running `$ python3 Screen.py` to test out the demo code? Does that fail on your setup?

The intention is to have a rotation of 0 [width, height] or 90 [height, width] for calculating the layout and developing the image. When displaying the image add 180 degrees yielding 0+180 = 180 or 90+180  = -90

txoof, 2021-07-07:
This now works as expected.

closed 2021-07-07 by txoof