txoof/epdlib #11: epd3in2 does not use the standard display method

txoof, 2021-04-11:
see [epd_display issue #2](https://github.com/txoof/epd_display/issues/2)

This may be resolved by the  [grayscale branch](https://github.com/txoof/epdlib/tree/grayscale)

robweber, 2021-04-29:
Found this while helping a user with another issue over on the [SlowMovie project](https://github.com/TomWhitwell/SlowMovie) (an EPD based slow video player). Wanted to point out a library we've been working on that abstracts EPD devices into a [common lib](https://github.com/robweber/omni-epd). Right now it handles pretty much all the waveshare models plus the Inky line of displays. Not sure if it's useful to you or not but it would be a way of supporting multiple display options without having to code around individual types in your project.

txoof, 2021-05-01:
@robweber THANK YOU! I started to write one of my own, but got frustrated with the **terrible** documentation over at WaveSahre. 

I'm super excited to try this out and see if I can port my layout library against this project.

robweber, 2021-05-01:
Agreed, there were so many places in the Waveshare lib we justed asked "why?". If you have any ideas for improvements feel free to toss them out.

txoof, 2021-07-05:
epd3in2 and epd1in02 screens are not supported - [see](https://github.com/txoof/epd_display/tree/new_epdlib#supported-screens)

https://github.com/txoof/epd_display/issues/2

closed 2021-07-05 by txoof