txoof/epdlib #12: Screen() object hangs forever when trying to init IT8951

txoof, 2021-07-05:
When no IT8951 display is connected, the `epdlib.Screen(epd='HD')` hangs forever.

This is related to how the IT8951 module pulls info from the board.
```
IT8951/interface.py in update_system_info(self)
     97         Get information about the system, and store it in class attributes
     98         '''
---> 99         self.spi.write_cmd(Commands.GET_DEV_INFO)
    100         data = self.spi.read_data(20)
    101 
```

txoof, 2021-07-05:
Related to https://github.com/GregDMeyer/IT8951/issues/36

txoof, 2021-07-05:
consider wrapping this in a `signal` handler that has a timeout for response:
https://stackoverflow.com/questions/492519/timeout-on-a-function-call

aaronr8684, 2022-03-06:
@txoof This is something I could fix as part of the rewrite of the Screen class. I don't know enough about the it8951 screens though to know what a reasonable timeout would be. I can research it but what are your thoughts?

txoof, 2022-03-06:
@aaronr8684 
This is basically a quality-of-life issue for end users. 

Based on my experience, I think a timeout of 15 seconds would be appropriate. I've never had a properly functioning IT8951 take more than 2 seconds to init. 15 seconds should be more than enough time for everything to init if it's working properly.

aaronr8684, 2022-03-06:
Alright, I'll see about adding it in since we're adding a bunch of new screens. 

Also, I believe the HD screens should be using Omni using combined code if you wanted to test sometime this week. I don't have one to test, so I can't personally confirm.

Do I understand correctly that the "HD"/IT8951 screens are the same thing as SPI/USB/I80 on waveshare's website?

txoof, 2022-03-07:
Yeah. HD:IT8951:UART/SPI are all synonyms for the same thing. Though there are some oddball boards mixed in there that look like they use HDMI.

aaronr8684, 2022-03-07:
Yea I saw a few of the HDMI boards from Good Display (I think).

So does that mean the documentation listed on your various epd repos incorrectly say that the SPI/USB/I80 screens aren't supported?

txoof, 2022-03-07:
> So does that mean the documentation listed on your various epd repos incorrectly say that the SPI/USB/I80 screens aren't supported?

Definitely should indicate that EPDLib supports Waveshare HD screenns. Which repos look wrong? I'll update.

aaronr8684, 2022-03-07:
On epd_display and PaperPi (last line):
[image]

Also I believe this statement about the BCM driver in the epdlib readme is no longer true (based on the [v0.1.0 release notes](https://github.com/GregDMeyer/IT8951#updates-for-version-010) ):
[image]

txoof, 2022-03-07:
I'll tidy those. Thanks for catching.

txoof, 2022-05-05:
This should be resolved now via [I#36](https://github.com/GregDMeyer/IT8951/issues/36) being closed on the IT8951 driver. Need to verify logic.

closed 2024-02-25 by txoof