txoof/epdlib #20: Rotation of EPD display: Waveshare 6" (800x600), rotations dont match documentation

jhaake2308, 2021-11-01:
Hi, this is concerning the documented rotation options of EPD display: 
With my waveshare 6" (800x600px - https://www.waveshare.com/wiki/6inch_e-Paper_HAT), the rotations dont match the documentation. Here this is matching:
cable left : 0; cable right: 180;  cable top: 90; cable bottom: -90

rotation = 0:
[image]

rotation = 90:

[image]

If you need to run test-code against my setup, feel free to get in touch. If you need any more data, pictures, etc., also get in touch.

Thank you and BR
Julius

txoof, 2021-11-03:
@jhaake2308 ahhh. I understand the issue much better. Thanks for the photos.

The 6" screen is set up differently than the other it8951 screens. I'll update the documentation to match. 

Long-term the solution is a bit more complicated as t his screen is an exception.

The text blocks are actually doing exactly what they are supposed to do. The font & size you've chosen can't render in the space provided in the layout so epdlib displays as many full words as it can, then shows the ellipses glyph "..." To indicate the text overflows the space. The top left field is one word and too large in the x dimension to display so it just gives an ellipses 

I should document this better.

The TextBlock layout type tries to calculate the maximum size for text given the max lines value, x & y dimensions and font. You are simply running out of x space.

 There are a few solutions that you can mix ad match: 

- use a different font (try something narrow in the x dimension), 
- adjust the layout to make a landscape and portrait version that uses space differently, 
- shorten up the text strings, 
- increase the max lines value in the layout

txoof, 2021-11-03:
@jhaake2308 Thanks for the issue; I've updated [the docs to reflect this](https://github.com/txoof/epdlib#properties-5)

closed 2021-11-03 by txoof