txoof/epdlib #21: possible ignore of max_refresh = 30

jhaake2308, 2021-11-14:
As it seems, the config parameter max_refresh (which is set to 30 in my case in paperpi.conf) is ignored. Every four changes there is a hard refresh. See this, the last refresh is done right before the video starts:

https://user-images.githubusercontent.com/75842297/141686012-de415ca1-7b32-4afc-a816-8fadb6a4b7a4.mp4

I am using a daemonized setup, defining max_refresh in /etc/default/paperpi.ini

```
# CONFIG_VERSION=1
[main]
# waveshare display type use HD for IT8951 displays
display_type = HD
# required vcom value for IT8951 screens
vcom = -2.13
# maximum refresh between total screen clear for HD displays
max_refresh = 30
```

is this intentional or did I do this wrong?

jhaake2308, 2021-11-14:
https://user-images.githubusercontent.com/75842297/141686819-e7e0740d-a5b6-4b66-a748-a22d1a6947fd.mp4

txoof, 2021-11-15:
@jhaake2308 This looks like a PaperPi issue, not an epdlib problem. I've [moved the ticket](https://github.com/txoof/epd_display/issues/20) there and closed this one.

closed 2021-11-15 by txoof