txoof/epdlib #26: Consider migrating to Omni EPD

txoof, 2022-02-03:
Omni EPD looks more reliable than the WaveShare defaults
https://github.com/robweber/omni-epd

Does not yet support IT8951 based screens

aaronr8684, 2022-02-09:
@txoof This might be a better long term solution to the 3in7patch problem. It also would help add support for greyscale and colors on the non-HD screens which is something I was looking at trying to make work. I know this won't be a quick fix, but what can I do to help? I have a greyscale (3.7"), BW/Red (5.83"), and 7-Color (5.65") that could be used for testing.

txoof, 2022-02-09:
@aaronr8684 The more I think about this, the more I like this option.

It would be really great to add grayscale support for everything and 7 color support. It should be possible to add 2 color support as well. We'll have to dig into how omni epd handles the 2-color. I'd like to eventually pick up a 7-Color screen for testing and playing as well.

From what I recall, the waveshare approach is to pass two separate images, one for the black and one for the "color." That's not unreasonable, but bringing those features into PaperPi will be a challenge.

If you have a working solution with your patched version of PaperPi, maybe it would be nice to prioritize working on rewriting epdlib to support Omni EPD. What do you think?

aaronr8684, 2022-02-09:
Alright, do you want to create a branch for me to work off of since this is probably a pretty big rework. I'm currently working in the Omni repo to test and verify support for the 7-color (done) and 3-color (in-progress) screens and fix a few bugs I found. Once that is done, I can start working on this.

aaronr8684, 2022-02-10:
Was able to get the 3-color fix done for the omni-epd library (if they accept my PR ;) ). Should be ready to tackle bringing it over here.

txoof, 2022-02-10:
@aaronr8684 Wow! Great progress! I'll have some more time this weekend to think about this.

txoof, 2022-02-10:
@aaronr8684 I've created a branch (I#26_omniepd) where we can work and test without breaking stuff. I just need to make sure I manage the PRs correctly :) This is always a challenge for me.

I'm looking forward to seeing what you've done. I'm also thinking about picking up a 7 color display. I'm not quite sure what I'll do with it, but they look sooo good.

aaronr8684, 2022-02-10:
I'm no expert and definitely learning, so no judgement from me 😄 

I'll keep you posted

aaronr8684, 2022-02-18:
In progress - Should also cover #25 if everything goes well. I think the initial goal will be to use Omni for the non_hd screens and continue to use the internal logic for the hd screens. Maybe eventually we can extend the Omni module to include the IT8951 as a subclass. One thing at a time though.

aaronr8684, 2022-02-18:
Right now, I'm just slowing working on converting the epdlib screen class to omni_epd. I have it talking to omni_epd, but due to a bug in the library, the particular screen I am using doesn't work as intended (PR pending). The takeaway though is that I'm using Omni 😄. I think once I have a semi-stable build of epdlib (with bw support only), I'll work on bringing in the epd_display/PaperPi module to do an end to end test. Once we have a working version, then I want to go back and see about adding color support and non-waveshare screen support. I'll let you know when I have something to test. Starting from a known working condition and running frequent tests is making this progress pretty quickly.

aaronr8684, 2022-02-18:
Alright, I have a semi-stable state if you want to run some tests and try a few screens. Documenting for me as much as for you

Pre-Reqs
Omni-epd - `sudo pip3 install git+https://github.com/robweber/omni-epd.git#egg=omni-epd`
Waveshare - `sudo pip install -e "git+https://github.com/waveshare/e-Paper.git#egg=waveshare_epd&subdirectory=RaspberryPi_JetsonNano/python"`

Install Test Branch
EPDLib - `git clone -b I#26_omniepd https://github.com/aaronr8684/epdlib.git`

Run Test
`python3 epdlib/epdlib/Screen.py`

If you are running a tri-color display and it shows a solid color (red or yellow) during the test, you'll need to modify the Omni-EPD library code in `/usr/local/lib/python<ver>/dist-packages/omni_epd/displays/waveshare_display.py`

After line 22:
Add - `from PIL import Image`

On line 202ish (in the bw if block):
Change this line - `self._device.display(self._device.getbuffer(image), [255] * (int(self.width/8) * self.height))`

To these two lines:
```
img_white = Image.new('1', (self._device.height, self._device.width), 255)
self._device.display(self._device.getbuffer(image), self._device.getbuffer(img_white))
```

txoof, 2022-02-18:
So far so good. I've tested your code against the following with no trouble:

- [x] epd5in83
- [x] epd2in7
- [x] epd2in7b

txoof, 2022-02-19:
Looks like Omni EPD merged in your PR. Does this resolve the issues you mentioned above with two color screens?

When i tested the latest OmniEPD, alongside your I#26 branch everything worked great.

aaronr8684, 2022-02-19:
He hasn't yet. The change only affects the tri-color displays.

aaronr8684, 2022-02-23:
Looks like we'll be getting IT8951 support soon on the Omni library. I'll keep that in mind as I was just planning on handling your implementation separately from the non-HD screens.

txoof, 2022-02-23:
That's really good news. It will make the logic in epdlib much cleaner.

I'll have to start looking at what changes this will mean for PaperPi. It won't be a huge change, but the configuration files will need some work.

The layouts logic in epdlib will also need some updating to accommodate screens that support grayscale and maybe even 3 color.

aaronr8684, 2022-02-28:
@txoof Do you have v1 or v2 of the 2in7b screen?

txoof, 2022-02-28:
It's pretty old. V1, i think...no obvious markings indicating that it is V2.

It's labeled as 2.7 (b) SKU 13357

aaronr8684, 2022-03-09:
I would say that once epdlib is versioned to v0.6 to include bw support via Omni, we could close this issue and open a new issue for grey and color support

txoof, 2022-03-10:
@aaronr8684 

I merged your PR into the Omni_EPD branch and tested it on the following: it8951 9" screen, ws epd5in83 and a ws epd2in7. The first two were a great success and had no problems.

The epd2in7 didn't display anything. I think the hardware is suspect rather than the code. I'll have to do some more testing to narrow down the problem to the second-hand pi or the epd screen.

I'll do some more validation over the next few days when I have some more time and test out some grayscale and try to figure out what's wrong with the 2in7b screen. 

Once that's done, we can close this issue and move to V0.6 on PyPi -- that's exciting!

aaronr8684, 2022-03-10:
@txoof There is a good chance that I broke greyscale support in the migration. I'm not too worried as that's what I'm going to be working on next anyway when I add that and color support, but it would be worth tracking at least.

aaronr8684, 2022-03-17:
Alright, I think I'm ready to tackle grey and color support in EPDLib. Do you want to do a merge into main first to get a stable v0.6 and then we can add the extra functionality into a v0.6.1 (or something similar)?

txoof, 2022-03-17:
doh. Should have started here. I just set up Paperpi to track the omni_epd branch in epdlib. 

I think I'd rather merge the omni_epd branch into the development branch and then set Paperpi up to track the devel branch. This keeps the "master" branch at 0.5 until we have a stable 0.6 version that we're satisfied with that has, at minimum, feature parity with 0.5.

aaronr8684, 2022-03-17:
Alright, fair enough. So installing PaperPi with the official installer should bring in the development branch from EPDLib, right?

txoof, 2023-06-20:
Decided not to use OMNI EPD and manually added color support.