txoof/epdlib #33: make option to flip horizontal/vertical

txoof, 2022-03-27:
Some WaveShare displays display all content mirrored over the X or Y axis for ??? REASONS ???. Add an option to mirror content along either of those axes.

aaronr8684, 2022-03-27:
So this is already an option in OmniEPD. Do we want to fix this for EPD_Display or just make sure it works for PaperPi?

https://github.com/robweber/omni-epd#advanced-epd-control

txoof, 2022-03-28:
I don't see much win in adding this to epd_display; it would mean a refactor to the main branch of epdlib and then changes in epd_display including the config file.

aaronr8684, 2022-03-28:
That was pretty much my thought as well. I figured, we'd kinda live with the limitations of epd_display until PaperPi is ready and then add any missing functionality to the new version if it's missing.

txoof, 2022-05-01:
@txoof 
went ahead and added this in [this PR](https://github.com/txoof/epdlib/pull/36) but now I'm reconsidering how I implemented this.

Maybe a good idea to rewind those PRs and just do it through OmniEPD.

txoof, 2022-05-03:
Pulled back the original PR and switched to using the IT8951 option. #37 

I'd like to use the OmniEPD feature when we get there, but I've had several requests for including this option now. As Omni doesn't support grayscale, I think this is worth merging in to the master branch.

txoof, 2022-05-03:
I have this running on a display right now; I'll see if run into any problems over the next few days and then merge this in.

closed 2023-12-20 by txoof