txoof/epdlib #34: cannot install inky/omni-epd in pipenv

txoof, 2022-04-30:
@aaronr8684 

I'm struggling to get omni-epd to install in my pipenv. Can you give any pointers for how you made this work? It keeps failing on numpy when it tries to install dependencies for _inky_. This is on a Buster32 with python 3.7. 

I've tried everything I can think of, but I can't get _numpy_ to install properly within the venv. It eventually always errors out with this:
```
      error: Command "arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/loops_unary_fp.dispatch.neon.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/loops_arithmetic.dispatch.neon.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/loops_unary_fp.dispatch.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/loops_arithm_fp.dispatch.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/loops_arithmetic.dispatch.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/loops_trigonometric.dispatch.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/loops_exponent_log.dispatch.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/loops_trigonometric.dispatch.neon_vfpv4.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/abstractdtypes.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/alloc.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/arrayobject.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/multiarray/arraytypes.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/array_coercion.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/array_method.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/array_assign_scalar.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/array_assign_array.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/arrayfunction_override.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/buffer.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/calculation.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/compiled_base.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/common.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/common_dtype.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/convert.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/convert_datatype.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/conversion_utils.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/ctors.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/datetime.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/datetime_strings.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/datetime_busday.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/datetime_busdaycal.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/descriptor.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/dtypemeta.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/dragon4.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/dtype_transfer.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/multiarray/einsum.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/multiarray/einsum_sumprod.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/flagsobject.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/getset.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/hashdescr.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/item_selection.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/iterators.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/legacy_dtype_implementation.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/multiarray/lowlevel_strided_loops.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/mapping.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/methods.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/multiarraymodule.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/multiarray/nditer_templ.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/nditer_api.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/nditer_constr.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/nditer_pywrap.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/number.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/refcount.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/sequence.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/shape.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/scalarapi.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/multiarray/scalartypes.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/strfuncs.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/temp_elide.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/typeinfo.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/usertypes.o build/temp.linux-armv7l-3.7/numpy/core/src/multiarray/vdot.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/npysort/quicksort.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/npysort/mergesort.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/npysort/timsort.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/npysort/heapsort.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/npysort/radixsort.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/npysort/selection.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/npysort/binsearch.o build/temp.linux-armv7l-3.7/numpy/core/src/umath/umathmodule.o build/temp.linux-armv7l-3.7/numpy/core/src/umath/reduction.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/loops.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/matmul.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/clip.o build/temp.linux-armv7l-3.7/numpy/core/src/umath/ufunc_object.o build/temp.linux-armv7l-3.7/numpy/core/src/umath/extobj.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/umath/scalarmath.o build/temp.linux-armv7l-3.7/numpy/core/src/umath/ufunc_type_resolution.o build/temp.linux-armv7l-3.7/numpy/core/src/umath/override.o build/temp.linux-armv7l-3.7/numpy/core/src/common/array_assign.o build/temp.linux-armv7l-3.7/numpy/core/src/common/mem_overlap.o build/temp.linux-armv7l-3.7/numpy/core/src/common/npy_argparse.o build/temp.linux-armv7l-3.7/numpy/core/src/common/npy_longdouble.o build/temp.linux-armv7l-3.7/numpy/core/src/common/ucsnarrow.o build/temp.linux-armv7l-3.7/numpy/core/src/common/ufunc_override.o build/temp.linux-armv7l-3.7/numpy/core/src/common/numpyos.o build/temp.linux-armv7l-3.7/build/src.linux-armv7l-3.7/numpy/core/src/common/npy_cpu_features.o build/temp.linux-armv7l-3.7/numpy/core/src/common/cblasfuncs.o build/temp.linux-armv7l-3.7/numpy/core/src/common/python_xerbla.o -L/usr/lib/arm-linux-gnueabihf -L/usr/lib -Lbuild/temp.linux-armv7l-3.7 -lnpymath -lf77blas -lcblas -latlas -lf77blas -lcblas -lm -o build/lib.linux-armv7l-3.7/numpy/core/_multiarray_umath.cpython-37m-arm-linux-gnueabihf.so" failed with exit status 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for numpy
ERROR: Could not build wheels for numpy, which is required to install pyproject.toml-based projects
```
https://github.com/txoof/epdlib/blob/1d2c4f0f582a4039eb7e3740ed4f6ecf1022f02f/epdlib/Screen.py#L16

aaronr8684, 2022-05-01:
Man, idk...let me take a look. I remember having some issues with that dependency but I don't recall how I got it working. Let me see how a fresh build does. Are you installing it fresh or on top of something? 

Unrelated, I found a CM4 at retail so I plan to get that up and running when they arrive.

Sorry I've been MIA lately. My job has been really busy lately and I was working on another project (home automation). I should start to have some more availability to continue working on PaperPi (and related) soon.

txoof, 2022-05-01:
@aaronr8684 
It was a pretty fresh install with nothing special changed. The debian python-numpy module works just fine, but I suspect that's just a binary package with all the wheels already compiled. 

This problem looks the compiler is barfing on ???? something ???? when trying to build the wheels.

No sweat on progress. I've been stupid busy at work and had zero brain space for anything other than quietly reading and listening to the soothing sounds of felt. 

Thanks for looking into this. I'll give it a go under Bullseye and see if I can get it to build there.

txoof, 2022-05-01:
@aaronr8684 So it looks like this is a problem with Buster32. It installs flawlessly on Bullseye32. 

For the moment, I guess we can just set the install requirements on the development branch to Bullseye32. Nothing wrong with that.

closed 2024-02-25 by txoof