txoof/epdlib #50: AttributeError on line 341 in Screen.py

matthewvinton, 2023-02-28:
Hey,

My second issue I do not know how to resolve, as I am not a programmer. The following line 341 in Screen.py gives an AttributeError complaining about getbuffer:

`self.buffer_no_image = self.epd.getbuffer(self.blank_image())`

I commented out the if block to get past the issue, as I am not sure what the true problem is.

txoof, 2023-02-28:
Shoot. I think I have a fix for that too. For the moment, I'll pin PaperPi to a previous version of epdlib.

txoof, 2023-02-28:
Thanks for catching these! I've already merged this change and #49 into the development branch. 

I've pinned the Pipfile in the stable version of PaperPi to version 0.5 of epdlib until I resolve all of these problems. I'm still getting the hang of managing packages and dependencies between projects.

txoof, 2023-02-28:
@matthewvinton 

Would you mind running the installer again and checking if everything works properly?

I just tested on my setup and it looks good, but another verification would be great.

matthewvinton, 2023-02-28:
No problem - I uninstalled, deleted, re-pulled from the main branch, and reinstalled. Unfortunately I had to make the same two modifications. Did I use the correct branch?

`git clone https://github.com/txoof/PaperPi.git`

txoof, 2023-03-01:
@matthewvinton 

Thanks again for checking. I see my error now. I misunderstood how pinning works in Pipfiles. The short version is I messed up and, I think I've fixed it. 

I had `epdlib = "0.5.2.1"`and it should read `epdlib = {version="==0.5.2.1"}`

If you pull the latest version of the repo and run the installer, or simply run the remote installer script it should fix the problem and pull against the 0.5 version of epdlib.

You can confirm that the proper version is installed with
```shell
$ cat /usr/local/paperpi/.venv/lib/python3.9/site-packages/epdlib/version.py

__version__="0.5.2.1"
```

closed 2024-02-25 by txoof