txoof/epdlib #65: Unexpected padding with right-aligned text in TextBlock

tristanhaeuser, 2024-03-12:
I'm working on a Waveshare 6'' HD EPD with IT8951 controller. 
When attempting to set up a right-aligned text within a text block I encounter unexpected spacing from the right border. Despite setting no padding and experimenting with various available properties, the issue persists. Additionally, I've noticed that even with left-aligned text, there's an invisible padding that restricts the utilization of the entire screen. This padding is problematic as it prevents optimal use of the display area.
Is this expected behaviour to have a safety margin on the right side or a bug somewhere in the library?

[image]

```
Section = {
    'text': {
        'type': 'TextBlock',
        'max_lines': 25,
        'align': 'right',
        'padding': 0,
        'width': 1,
        'height': 1,
        'abs_coordinates': (0, 0),
        'hcenter': False,
        'vcenter': True,
        'relative': None,
        'font': sf_font,
        'font_size': None,
        'inverse': False,
        'text_wrap': True,
        'fill': 'White',
        'bkground': 'Black',
        'mode': 'L'
    }
}
```

txoof, 2024-03-13:
You've probably found a bug. That definitely doesn't look right.
Thanks for providing the layout and font. 

Let me take a poke at this and see if there's something I can do to fix it. 

Would you mind sharing an image for the left-aligned issue as well. 

Just to be certain, is the resolution of your [screen](https://www.waveshare.com/wiki/6inch_HD_e-Paper_HAT) 1448 X 1072?

tristanhaeuser, 2024-03-13:
Yes, that's correct the resolution is 1448 x 1072. 
Here is an image for the left-aligned issue:

[image]

I think the bug is not font-specific. I used a variety of fonts while developing and it doesn't seem like it would make any difference.

txoof, 2024-03-13:
@tristanhaeuser I'll see if I can fix the justification issue and at least get right justification to shove over to the right side and leave "useless" space on the left instead.

As for distributing text better, I'm open to suggestions. I really struggled to get a solution that worked well for most `TextBlock` sizes. The calculations happen in `[Block.TextBlock](https://github.com/txoof/epdlib/blob/68e9e9eaa6c2ab1084f4fb128488b761f42e9ce4/epdlib/Block.py#L965-L991)`. If you have any ideas how to make this a bit better, I'd love to have some other brains working on it.

Ideally, it would be a bit more dynamic and flow text with the logic of a soft-wrap, but I haven't really thought much about how to do that. I was focused on using the `textwrap` library.

txoof, 2024-03-13:
looks like `TextBlock` needs an additional case for right aligned text somewhere around [l1087](https://github.com/txoof/epdlib/blob/68e9e9eaa6c2ab1084f4fb128488b761f42e9ce4/epdlib/Block.py#L1087-L1093).

It's too late tonight to figure out the math, but this is the gist:
```
if self.align == 'ralign'
   paste_x = <<insert math here for shoving text to right border>>>
   logging.debug('pasting raligned text on the right side')
```

txoof, 2024-03-15:
@tristanhaeuser 
I've done some research and I can't find a good alternative to textwrap other than trying to do it from the ground up. 

If you have any ideas, I'd love to hear them. I'm going to dig into your right justification bug shortly and hopefully have a solution this weekend.

txoof, 2024-03-15:
@tristanhaeuser well, that was easier than I thought!

I've got a fix ready to go, but I'd appreciate it if you could test it out. The output is below. 

You can install the issue-branch with:

`pip install git+https://github.com/txoof/epdlib/epdlib.git@I65_alignment`

[image]

tristanhaeuser, 2024-03-17:
@txoof  Wow, thank you for coming up with a fix so quickly. I'm currently very busy, but I try it out as soon as possible. 
I let you know when I tested it. Should I close this issue?

txoof, 2024-03-17:
No problem. I hope this resolves your problem. The fix was earlier than I thought.

Leave it open until you've tested. I'll close it when I merge the update.

tristanhaeuser, 2024-03-18:
@txoof I installed the branch with the fix. As far as I can tell everything works now as expected.  : )

txoof, 2024-03-18:
Thanks for checking. I'll merge it in when I get a minute and publish to PyPi. 

Don't forget to update your requirements.txt!

txoof, 2024-03-20:
@tristanhaeuser the v0.6.5 is up on pypi. Happy hacking!

If you find any other issues with this, feel free to reopen the ticket.

closed 2024-03-20 by txoof